ioBroker/modbus-templates #5: tsv Datei erstellen

Schnubby666, 2023-01-30:
Ich habe gerade ein Drehstromzähler mit Modus RTU in den ioBroker integeriert. Wie erstelle ich aus dem abgetippten Register eine tsv Datei?

Toki99, 2023-01-31:
Du muss beim Holding Register auf die beiden Pfeile gehen und dann per copy & paste den Text kopieren

Schnubby666, 2023-01-31:
Ich habe in dem Fall nur ein Eingangsregister.
Wenn ich das Kopiere, habe ich ja nur ein iwas in der Zwischenablage. Wo kann ich dass dann einfügen um es als .tsv abspeichern zu können?

Triton-KS, 2023-02-18:
Servus,
die Pfeile gibt es auch im Eingangsregister. Wei du schon schreibst einfach in die Zwischenablage kopieren und dann in einem "Text"-Editor deiner Wahl einfügen und entsprechend speichern.
So hat mein mir der Export und Import funktioniet.

closed 2026-10-25 by GermanBluefox