ioBroker/modbus-templates #6: KACO blueplanet 10.0 TL3 & KACO blueplanet 5.0 TL3

Schimi1983, 2023-02-09:
[image]

Holding Register (gibt noch viel mehr, mir haben diese gereicht)
`_address	name	description	unit	type	len	factor	offset	formula	role	room	poll	wp	cw	isScale
40144	5kW_Erzeugung	Aktuelle Gesamt Erzeugung	W	floatbe	2	1	0		level		true	false	false	false
40154	5kW_GesamtErtrag	WattHours	kWh	floatbe	2	0.001	0		level		true	false	false	false
40162	5kW_Temperatur	Temperatur	C	floatbe	2	1	0		level		true	false	false	false`

Das "SPEC_HelpTool_Client&Manual.zip"
[SPEC_HelpTool_Client&Manual.zip](https://github.com/ioBroker/modbus-templates/files/10699089/SPEC_HelpTool_Client.Manual.zip)

Weitere Modbus Adressen (SPEC_SMI_INT_bp_3.0-10.0TL3_V5.53):
[SPEC_SMI_INT_bp_3.0-10.0TL3_V5.53.zip](https://github.com/ioBroker/modbus-templates/files/10699092/SPEC_SMI_INT_bp_3.0-10.0TL3_V5.53.zip)

KronenbergSmarthome, 2023-10-26:
Ich werde demnächst einen KACO blueplanet 8.0 NX3 M2 bei mir Inbetrieb nehmen. Er unterstützt Modbus TCP. Wird das Modbus Protokoll per WLAN gesendet oder funktioniert es über eine der beiden RJ45 Anschlüsse? Laut der Anleitung liefern diese beiden Anschlüsse aber nur ein RS485 Protokoll. Verstehe den Zusammenhang mit RS485 und Modbus TCP nicht so ganz. Und kennt jemand einen Adapter für ioBroker mit dem ich die Daten vom Kaco Wechselrichter auslesen kann?

Schimi1983, 2023-10-26:
Probier den modbus Adapter....

die blueplanet TL3 geben modbus über LAN aus... dann vermute ich mal ist es einen versuch über WLAN wert....

einfach probieren.... vielleicht gehen ja die modbus adressen vom TL3

KronenbergSmarthome, 2023-10-26:
Jo danke, werde ich mal versuchen. Verstehe es so, dass Modbus TCP auf Basis von einem TCP/IP funktioniert, quasi die gleiche Netzwerktechnik wie man die normalerweise zu Hause verwendet oder werfe ich da was durcheinander? Habe bis jetzt noch keine vernünftigen Informationen über die Integration von Kaco blueplanet in ioBroker gefunden. Scheint echt nicht so einfach zu sein. Bin jetzt schon am zweifeln ob mein Installateur nicht besser einen anderen WR hätte nehmen sollen, oder ich einen vorgeschlagen hätte von dem ich weiß dass es einen Adapter für ioBroker gibt.

Schimi1983, 2023-11-05:
habe vergessen zu antworten....

also Modbus TCP läuft (zumindest bei den blueplanets TL3) ganz normal über das Netzwerkkabel womit die Wechselrichter mit dem Netzwerk verbunden sind.....

Läuft bei mir schon seit Jahren ohne probleme....
Was man bedenken muss, das immer nur eine verbindung aktiv sein kann... also nicht mehrere Modbus verbindungen mit einem Wechselrichter...

KronenbergSmarthome, 2023-11-05:
Danke für die Antwort. Meine KACO WR ist blueplanet 8.0 NX3 M2 und da scheint es vollkommen anders zu sein. Ich verstehe die dämliche Anleitung von KACO nicht. Unter technische Daten steht folgendes:
[image]
Was denn nun? RS485 und dann Modbus TCP? TCP nutzt doch Netzwerktechnik oder?
Hier ist der Anschluss dafür beschrieben und kapiere überhaupt nicht mehr.
[image]
[image]
Verstehe nicht, wie man in der heutigen Zeit alles noch so kompliziert machen muss? Kann man nicht einfach den WR ins Netzt hängen und das Smartmeter auch und darüber ne Kommunikation aufbauen, und vom Smartmeter kann ich mir dann meine Daten abholen? Ich bin selbst Elektroniker und SPS Programmierer und meiner Meinung nach nicht ganz auf den Kopf gefallen, aber das ist mir echt zu hoch, warum man es so kompliziert machen muss.

fefaul, 2024-03-23:
#32

robschii, 2024-07-14:
> Danke für die Antwort. Meine KACO WR ist blueplanet 8.0 NX3 M2 und da scheint es vollkommen anders zu sein. Ich verstehe die dämliche Anleitung von KACO nicht. Unter technische Daten steht folgendes: ![image](https://private-user-images.githubusercontent.com/93072708/280534415-cb8f4bcb-08fa-42a7-8739-4e44f5bb9534.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA5ODQ4NzcsIm5iZiI6MTcyMDk4NDU3NywicGF0aCI6Ii85MzA3MjcwOC8yODA1MzQ0MTUtY2I4ZjRiY2ItMDhmYS00MmE3LTg3MzktNGU0NGY1YmI5NTM0LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxNFQxOTE2MTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zM2IyYTY1ZmRlYmY5MDI1OTQ4MTU5ODdjMzA0ZWIxYjc4YjUzMGRlMDdmNjYxZWZkMWRiOWVhNTdhODk2NDhmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.19nt7rC71bstQXkUK9NOUebFZFi5emi_nZpXPQVgizk) Was denn nun? RS485 und dann Modbus TCP? TCP nutzt doch Netzwerktechnik oder? Hier ist der Anschluss dafür beschrieben und kapiere überhaupt nicht mehr. ![image](https://private-user-images.githubusercontent.com/93072708/280534580-cd31d318-90f4-451f-b62c-1433c36e3dbb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA5ODQ4NzcsIm5iZiI6MTcyMDk4NDU3NywicGF0aCI6Ii85MzA3MjcwOC8yODA1MzQ1ODAtY2QzMWQzMTgtOTBmNC00NTFmLWI2MmMtMTQzM2MzNmUzZGJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE0VDE5MTYxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY3MTYyNjY2YWRlMmQ5ZjUxODljMmVmZGIyN2U0OTAyMWRmNjVjNDY3ZDkyMzBlNDkxODZlZjdmYjZjNzlkNGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._v1PvJzMWpJw-ozDfIW-iRqOdIvbb3vwnnwReyiRE0A) ![image](https://private-user-images.githubusercontent.com/93072708/280534593-eb01e1c8-0288-4a37-8319-32f7d18aee5c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA5ODQ4NzcsIm5iZiI6MTcyMDk4NDU3NywicGF0aCI6Ii85MzA3MjcwOC8yODA1MzQ1OTMtZWIwMWUxYzgtMDI4OC00YTM3LTgzMTktMzJmN2QxOGFlZTVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE0VDE5MTYxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQyNjZhNDBmNjY5M2E3OWM1NjkyNWE4N2YxNmEyZGFkYmQwYjUzNmIzYjc0YTFhNWVmNmQ3M2UzNGU1Y2YzNDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qP8yNOKVxyaNcpDFB8WMB57S6Orl0npQt-rLuw-mmto) Verstehe nicht, wie man in der heutigen Zeit alles noch so kompliziert machen muss? Kann man nicht einfach den WR ins Netzt hängen und das Smartmeter auch und darüber ne Kommunikation aufbauen, und vom Smartmeter kann ich mir dann meine Daten abholen? Ich bin selbst Elektroniker und SPS Programmierer und meiner Meinung nach nicht ganz auf den Kopf gefallen, aber das ist mir echt zu hoch, warum man es so kompliziert machen muss. Also müsste ich mir erstmal nen Adapter besorgen oder basteln, damit ich von dem RJ45 Stecker die 2 Adern rausholen kann und den dann am Smartmeter anschließen, aber wie bekomme ich dann die Daten vom Smartmeter? Brauche ich dafür wieder nen Adapter von Modbus auf TCP/IP um es in iobroker mit dem Modbus Adapter einlesen zu können?
> 
> Wäre echt astrein wenn mir jemand dieses Thema näher erklären könnte. Vielleicht ist es auch gar nicht so kompliziert und stehe einfach nur auf dem Schlauch 😂
> 
> PS: Das ist das Smartmeter welches dazu gehört und das das SunSpec Protokoll unterstützt was der KACO liefert. ![image](https://private-user-images.githubusercontent.com/93072708/280535587-2c375ecf-d577-4cb7-87a2-78632896ddd3.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA5ODQ4NzcsIm5iZiI6MTcyMDk4NDU3NywicGF0aCI6Ii85MzA3MjcwOC8yODA1MzU1ODctMmMzNzVlY2YtZDU3Ny00Y2I3LTg3YTItNzg2MzI4OTZkZGQzLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxNFQxOTE2MTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMmQyZWQwYmNhMzQwNmI0ZjM0MDlkYzY1Y2YzNWZkOTU4MzFjZTczM2EyYjRjZDgyYzkzZmU4YTU2ZTVlYjUwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.h1nuUNidyobJcxfyuTL6b9ZRa-n5J8B4avrhO8bt3yU)

nabend, ich habe auch den blueplanet 8.0 und versuche auch über die IP ( Gerät hängt im Wlan) den Modbus abzugreifen.

@KronenbergSmarthome  hast du es hinbekommen und wie?

der Varta Speicher läuft bei mir bereits über modbus

KronenbergSmarthome, 2024-07-17:
> nabend, ich habe auch den blueplanet 8.0 und versuche auch über die IP ( Gerät hängt im Wlan) den Modbus abzugreifen.
>
> @KronenbergSmarthome hast du es hinbekommen und wie?
>
> der Varta Speicher läuft [...]

Hi @Schimi1983 ,
ja ich habe es hinbekommen. In der Kaco App habe ich unter Überwachung & Steuerung die Kommunikation auf Modbus TCP IP gestellt. Kaco Blueplanet Web funktioniert danach nicht mehr, aber das ist eh einer der schlechtesten Visualisierungen und Weboberflächen für Photovoltaik, die ich je gesehen habe. Danach konnte ich im ioBroker per Modus.0 Instanz mit dem Wechselrichter kommunizieren. Die Einstellungen dafür sehen folgendermaßen aus.
[image]
[image]
Danach habe ich an meinen Eastron Smartmeter SDM630 eine Waveshare RS485 to Ethernet Adapter angeschlossen und diesen auch auf Modbus TCP gestellt. Diesen lese ich auch im ioBroker mit einer zweiten Modbus.1 Instanz ein.
[image]
[image]
[image]

Diese Daten (Nur die ich benötige) logge ich dann per InfluxDB Instanz in zwei verschiedenen Instanzen ab. Die Daten vom Smartmeter habe ich in einer InfluxDB.1 Instanz abgelegt, wo die Vorhaltezeit nur auf einen Monat eingestellt ist, da es sonst eine zu große Datenmenge werden würde. Für die Ertrags- und Verbrauchdaten benutze ich den Sourceanalytics Adapter der mir die Erzeugung und den Verbrauch ganz einfach berechnet, protokolliert und aufbereitet. Die Daten logge ich dann per InfluxDB.0 Instanz (Diese Daten werden immer vorgehalten).

Visualisiert habe ich die Daten von beiden InfluxDB´s dann per Grafana und zusätzlich noch mit dem ioBroker Adapter iQontrol (Diese benutzte ich auch für sämtliche anderen Steuerungen in meinem Smarthome)
Hier ein paar Beispiele aus Grafana:
[image]
[image]
[image]
Hier ein paar Beispiele aus iQontrol:
[image]
[image]

ioBroker, Grafana und InfluxDB laufen bei mir auf einer QNAP TS251A in einem Container.

closed 2026-10-25 by GermanBluefox