ioBroker/modbus-templates #9: Weitere Odner erstellen oder Templates) zur Vefügung stellen.

Triton-KS, 2023-02-18:
Hallo Matthias,

nach deinen Kursen und Videos konnte ich jetzt auch meinen Energiezähler über Modbus auslesen.
Es ist ein 3 Phasen Energiezähler der Firma Finder.
"Typ 7M.38 - Bidirektionaler Multifunktionszähler"
Teilweise sind die Adressen auch vom Typ 7M.24 (Einphasen-Energiezähler) verwendbar.

Wie kann ich der ioBroker-Community diese Templates zur Verfügung stellen?
Kannst du möchtest du neben den Wechselrichtern einen neuen Ordner für Energiezähler anlegen und ich schicke hier die Daten oder wie wäre der richtige Weg?

Apollon77, 2023-02-19:
Hi, das kannst Du selbst tun, hier auf der GitHub Webseite. gehe oben aif "File anlegen" und gib den Pfad an - fehlende Verzeichnisse werden dann automatisch angelegt

Triton-KS, 2023-02-19:
Danke für die Rückmeldung. 
Habe ich jetzt bestimmt an der falschen Stelle oder falsche Art gemacht. 
Können wenn nötg die Fehler korregiert werden?
Gedacht war nach dem Ordner modbus-templates einen Ordner für Energiezähler und dann einen Unterordner für die Finderzähler anzulegen

Danke

Triton-KS, 2023-02-20:
[README.md.txt](https://github.com/ioBroker/modbus-templates/files/10786590/README.md.txt)
[input-registers.txt](https://github.com/ioBroker/modbus-templates/files/10786593/input-registers.txt)

Servus,
Ich habe es nochmal direkt versucht aber nicht sicher ob es funktioniert hat.
Daher jetzt per Dateien. Leider kann ich keine TSV-Datei direkt hochladen. Daher die Endung TXT

closed 2023-02-22 by Triton-KS