ioBroker/modbus-templates #15: Register for Solaredge Home Battery

breznsoizer, 2023-08-30:
Thank you for providing the templates. where can I find the registers for the Solaredge Home Battery 48v

JoGu01, 2023-08-30:
Hi @breznsoizer ,
i am using the attached Holding Registers for mine.
[solaredge_battery.zip](https://github.com/ioBroker/modbus-templates/files/12476300/solaredge_battery.zip)
Regards,
Jörg

breznsoizer, 2023-08-31:
Hi @JoGu01,
Thank you for your holding registers.
They work for me too. Great!
Regards,
Breznsoizer

UweSauter, 2024-12-19:
Have a look at https://github.com/ioBroker/ioBroker.modbus/issues/377

Apollon77, 2025-01-02:
Could someone do a PR? with exported values?

GermanBluefox, 2026-10-25:
Die Register für die SolarEdge Home Battery sind inzwischen im Repo (PV-Wechselrichter/SolarEdge, hinzugefügt mit #35 und #41).

closed 2026-10-25 by GermanBluefox