ioBroker/modbus-templates #20: Register für Wallbox DeltaACMax per WLAN

heitec11, 2024-01-06:
Folgenden Aufbau habe ich verwendet:

ESP8266 mit ESPeasy als Device: "Communication - Serial Server"
TTL auf RS485 Modul mit RX/TX an D1/D2 (nicht DI,DE,RE,RO)
iobroker mit modbus Adapter im Modus "RTU over TCP"

Hier meine aktuelle TSV:
[DeltaACMax.tsv.txt](https://github.com/ioBroker/modbus-templates/files/13848939/DeltaACMax.tsv.txt)

Folgende Einstellungen:
[image]
[image]

Als Ergebnis:
[image]

Nun suche ich einen Anschluss innerhalb der Wallbox, wo ich 5V Versorgungsspannung abgreifen kann.

heitec11, 2024-01-06:
Schaltplan ESP:
[image]

Webiga, 2025-01-20:
Hallo heitec11,
dies klingt schon einmal gut, und wie genau ist ESPEasy eingestellt - ich bekomme keinen Verbindung hin :(
Viele Grüße
Webiga

closed 2026-10-25 by GermanBluefox