ioBroker/modbus-templates #33: KACO new energy Inverter

MitchMitchel23, 2024-03-28:
Im Pullrequest #32 wurden die Modbusregister für die blueplanet 3.0 NX3 M2 – 20.0 NX3 M2 Serie hinzugefügt. Diese enhält allerdings mehrere Fehler:

- Bei den Werten 40305 und 40328 sind falsche Faktoren eingetragen (1 statt 0.1).
- Die Werte 40084, 40086 und 40094 können nicht direkt verwendet werden, sondern hängen von den Skalierungsfaktoren in den Werten 40085, 40087 und 40096 ab und müssen berechnet werden.

Einen aktuellen Export für einen einzelnen Wechselrichter hänge ich an (umbennen von .txt in .tsv).
[holding-registers_blueplanet_nx3_m2.txt](https://github.com/ioBroker/modbus-templates/files/14791223/holding-registers_blueplanet_nx3_m2.txt)

robschii, 2024-07-14:
wie komm ich auf den modbus?

ip adresse (wlan stick)? und port 502?   oder ganz anders?

robschii, 2024-07-14:
@MitchMitchel23 

wie komm ich auf den modbus?

ip adresse (wlan stick)? und port 502?   oder ganz anders?

MitchMitchel23, 2024-07-15:
@robschii 
Verbindungsparameter: TCP
IP-Adresse und Port 502
Geräte ID: 3
"Aliases benutzen" und die anderen Einstellungen unter "Allgemeines" habe ich deaktiviert.

closed 2026-10-25 by GermanBluefox