ioBroker/modbus-templates #42: Solaredge: Type bei Zählerwert der importieren bzw. exportierten Batterie-Energie

ManfredHi, 2024-08-05:
Wenn ich bei meinem SolarEdge SE10K-RWS Wechselrichter  bei den Modbus-Registern für den Zähler der exportieren bzw. importierten Energie der Batterie bei Type "uint64le" einstelle, dann zählt der zähler sehr rasch hoch (bei jeder Wertänderung so um die 200 ca.). Wenn ich den Type "unit32sw" einstelle, dann wird immer nur um 1 kwh hochgezählt und das Endergebnis am Tag stimmt annähernd mit dem zusammen was die mySolaredge-App ausgiebt.

[image]

Was ist nun richtig? Oder ist der Wert kein kWh sondern ein Wh wert ?

UweSauter, 2024-12-19:
Das habe ich schon hier diskutiert: https://github.com/ioBroker/ioBroker.modbus/issues/377

GermanBluefox, 2026-11-12:
The template `holding-registers-S10KRWB-SE30K-Battery.tsv` reads the energy counters (0xE176 / 0xE17A) as `uint32sw` already. SolarEdge stores them as 64 bit with swapped words; `uint32sw` reads the two low words, which is correct for values below 2^32 Wh (about 4.3 GWh). See also ioBroker/ioBroker.modbus#377.

The addresses of this file were corrected in #86. Closing, please reopen if the values are still wrong.

closed 2026-11-12 by GermanBluefox